analogdevicesinc/TMC-API #1: u16 typedef multiple declaration

dzid26, 2019-11-19:
I am not sure if the problem is in Arduino core library or TMC_api, but in order to compile, I had to comment out line 34 in helpers/Types.h:
typedef uint8_t  u8;
// typedef uint16_t u16;
typedef uint32_t u32;

Error message otherwise:
> In file included from .pio\libdeps\nanoatmega328\TMC-API/tmc\helpers\Config.h:12:0,
                 from .pio\libdeps\nanoatmega328\TMC-API/tmc\helpers\API_Header.h:11,
                 from src\SPI_TMC.h:11,
                 from src\main.cpp:5:
.pio\libdeps\nanoatmega328\TMC-API/tmc\helpers\Types.h:34:18: error: conflicting declaration 'typedef uint16_t u16'
 typedef uint16_t u16;
                  ^
In file included from C:\Users\jareckir\.platformio\packages\framework-arduinoavr\cores\arduino/Arduino.h:233:0,
                 from src\main.cpp:1:
C:\Users\jareckir\.platformio\packages\framework-arduinoavr\cores\arduino/USBAPI.h:30:24: note: previous declaration as 'typedef short unsigned int u16'
 typedef unsigned short u16;
                        ^

Also, I needed to remove other ICs except one that I am using (TMC5160) because there were too many errors within them.

trinamic-LH, 2019-12-02:
Both our API and the Arduino USBAPI.h declare the u8/u16/u32 types - and do so differently. The fix for that is to remove one of them, which you already applied.

This is an issue we stumbled upon earlier too. Hence we started changing the entire API to only use the types as defined by stdint.h (uint8_t etc.). Due to backwards compatibility issues we have not removed the old declarations - if we were to remove them, any customer application code using our u8/u16/u32 definitions would break.

What we can do is hide the definitions behind a compile switch, e.g. `#define ENABLE_INTEGER_TYPEDEFS` and have a well documented place where this should be written/commented out depending on whether a customer wants us to define these types or not.

dzid26, 2019-12-02:
Your solution sounds like the best approach. 

Regarding other errors -  
I don't have too much experience with best practises with buildiing large projects, so it could be something wrong with my process. 
The compiler builds everything present in the folder by defualt. 
Here is what I get if I keep all the IC folders:
```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/nanoatmega328.html
PLATFORM: Atmel AVR 1.15.0 > Arduino Nano ATmega328
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 30KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.1.2
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <TMC-API> #116bf74
Building in release mode
Compiling .pio\build\nanoatmega328\src\main.cpp.o
Compiling .pio\build\nanoatmega328\lib0bc\TMC-API\tmc\ic\TMC2041\TMC2041.c.o
Compiling .pio\build\nanoatmega328\lib0bc\TMC-API\tmc\ic\TMC2130\TMC2130.c.o
Compiling .pio\build\nanoatmega328\lib0bc\TMC-API\tmc\ic\TMC2160\TMC2160.c.o
Compiling .pio\build\nanoatmega328\lib0bc\TMC-API\tmc\ic\TMC2208\TMC2208.c.o
Compiling .pio\build\nanoatmega328\lib0bc\TMC-API\tmc\ic\TMC2209\TMC2209.c.o
Compiling .pio\build\nanoatmega328\lib0bc\TMC-API\tmc\ic\TMC2224\TMC2224.c.o
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2041\TMC2041.c: In function 'tmc2041_writeDatagram':
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2041\TMC2041.c:19:18: warning: left shift count >= width of type [-Wshift-count-overflow]
  int value = (x1 << 24) | (x2 << 16) | (x3 << 8) | x4;
                  ^
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2041\TMC2041.c:19:31: warning: left shift count >= width of type [-Wshift-count-overflow]
Compiling .pio\build\nanoatmega328\lib0bc\TMC-API\tmc\ic\TMC2225\TMC2225.c.o
  int value = (x1 << 24) | (x2 << 16) | (x3 << 8) | x4;
                                ^

.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2041\TMC2041.c: In function 'tmc2041_readInt':
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2041\TMC2041.c:48:18: warning: left shift count >= width of type [-Wshift-count-overflow]
  return (data[1] << 24) | (data[2] << 16) | (data[3] << 8) | data[4];
                  ^
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2041\TMC2041.c:48:36: warning: left shift count >= width of type [-Wshift-count-overflow]
  return (data[1] << 24) | (data[2] << 16) | (data[3] << 8) | data[4];
                                     ^

.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2130\TMC2130.c: In function 'tmc2130_writeDatagram':
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2130\TMC2130.c:22:22: warning: left shift count >= width of type [-Wshift-count-overflow]
  int32_t value = (x1 << 24) | (x2 << 16) | (x3 << 8) | x4;
                      ^
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2130\TMC2130.c:22:35: warning: left shift count >= width of type [-Wshift-count-overflow]
  int32_t value = (x1 << 24) | (x2 << 16) | (x3 << 8) | x4;
                                   ^
..pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2130\TMC2130.c: In function 'tmc2130_readInt':

.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2130\TMC2130.c:53:18: warning: left shift count >= width of type [-Wshift-count-overflow]
  return (data[1] << 24) | (data[2] << 16) | (data[3] << 8) | data[4];
                  ^
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2130\TMC2130.c:53:36: warning: left shift count >= width of type [-Wshift-count-overflow]
  return (data[1] << 24) | (data[2] << 16) | (data[3] << 8) | data[4];
                                    ^
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2160\TMC2160.c: In function 'tmc2160_writeDatagram':
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2160\TMC2160.c:20:18: warning: left shift count >= width of type [-Wshift-count-overflow]
  int value = (x1 << 24) | (x2 << 16) | (x3 << 8) | x4;
                  ^
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2160\TMC2160.c:20:31: warning: left shift count >= width of type [-Wshift-count-overflow]
   int value = (x1 << 24) | (x2 << 16) | (x3 << 8) | x4;

                               ^
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2160\TMC2160.c: In function 'tmc2160_readInt':
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC2160\TMC2160.c:49:18: warning: left shift count >= width of type [-Wshift-count-overflow]
  return (data[1] << 24) | (data[2] << 16) | (data[3] << 8) | data[4];
[... 318 lines cut ...]

.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC4330\TMC4330.h:20:35: note: in expansion of macro 'FIELD_SET'
  (tmc4330_writeInt(tdef, address, FIELD_SET(tmc4330_readInt(tdef, address), mask, shift, value)))
                                   ^
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC4330\TMC4330.c:307:3: note: in expansion of macro 'TMC4330_FIELD_UPDATE'
   TMC4330_FIELD_UPDATE(tmc4330, TMC4330_ENC_IN_CONF, TMC4330_CL_CALIBRATION_EN_MASK, TMC4330_CL_CALIBRATION_EN_SHIFT, 0);
   ^
.pio\libdeps\nanoatmega328\TMC-API/tmc/helpers/Macros.h:33:36: warning: left shift count >= width of type [-Wshift-count-overflow]
  (((data) & (~(mask))) | (((value) << (shift)) & (mask)))
                                    ^
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC4330\TMC4330.h:20:35: note: in expansion of macro 'FIELD_SET'
  (tmc4330_writeInt(tdef, address, FIELD_SET(tmc4330_readInt(tdef, address), mask, shift, value)))
[... cut ...]
.pio\libdeps\nanoatmega328\TMC-API\tmc\ic\TMC4330\TMC4330.c:309:3: note: in expansion of macro 'TMC4330_FIELD_UPDATE'
   TMC4330_FIELD_UPDATE(tmc4330, TMC4330_ENC_IN_CONF, TMC4330_REGULATION_MODUS_MASK, TMC4330_REGULATION_MODUS_SHIFT, 1);
   ^
================================================================== [FAILED] Took 5.65 seconds ========
```

Some of the errors from above snippet:
conflicting types for 'ReadEncoder'
'**TMC423_ENC_CONF**_1' undeclared 
'**SPI_DEV_ENCODER**' undeclared
'**TRUE**' undeclared 
'**FALSE**' undeclared

trinamic-LH, 2019-12-02:
The warnings are due to the Arduino having a 16-bit int data type - `sizeof(int) == 2` is true on Arduino. While technically wrong, these should all work fine in practise.

The errors are all coming from the ICs TMC262, TMC424, TMC4210 and TMC429. These have not been modified in a while and are not actively used in our reference project, the TMC-EvalSystem - hence the issues with them. Removing them is the easiest way to deal with this for you, as you aren't using those.

trinamic-LH, 2020-06-10:
Fixed in 59c44a52fa82b7e7e2ea69812bfe0446c5adf8d4

To fix the multiple typedef errors/warnings, [uncomment the two new defines in Types.h](https://github.com/trinamic/TMC-API/blob/59c44a52fa82b7e7e2ea69812bfe0446c5adf8d4/tmc/helpers/Types.h#L8) or alternatively supply the defines through your build process. For example, when using a Makefile build, add the flags `-DTMC_SKIP_UINT_TYPEDEFS  -DTMC_SKIP_INT_TYPEDEFS ` to the compiler flags

closed 2020-06-10 by trinamic-LH