analogdevicesinc/TMC-API #2: is source for TMC2300 and TMC2300_EVAL available?

jancumps, 2020-01-25:
[There's a binary dowload](https://www.trinamic.com/fileadmin/assets/Products/Eval_Software/TMC2300-EVAL_Firmware_Landungsbruecke_v1.07.hex) of the firmware specific for TMC2300-EVAL. 

Can the source for the TMC2300 IC (registers) and evaluation board be added to github?

trinamic-LH, 2020-01-27:
The TMC2300 is currently not yet integrated into the TMC-EvalSystem project. As soon as this is done, the source code will be publically available.

Reason for the separate version were multiple challenges with booting up the chip safely in the face of legacy behaviour with the driverEnable pin and the ID detection mechanism. As a result the TMC2300 started in an undesired powered-on state before the ID detection mechanism could determine that a TMC2300 is connected to the Landungsbruecke. Recent improvements in the ID detection mechanism brought the detection time (and therefore undesired power state) low enough to be tolerable.

trinamic-LH, 2020-01-31:
Small update:
The source code should be published within the next week

trinamic-LH, 2020-02-07:
The TMC2300 source code is published.

Right now there is still an issue at startup - the motor receives current for ~23ms while the Landungsbruecke is still identifying the attached boards. If you are using a very small motor this may cause damage to it. To mitigate this, start the Landungsbruecke without having the power and/or motor connected.

closed 2020-02-07 by trinamic-LH