analogdevicesinc/TMC-API #3: Assumed datatype in CAST_Sn_TO_S32 in Macros.h

PMTaylor, 2020-02-16:
Hi there,
using Arduino, and the datatype the compiler assumes is too short (int?) for the operation if the shift is bigger than 16.

Seems to work as intended as 
"#define CAST_Sn_TO_S32(value, n) ((value) | (((value) & (1L<<((n)-1)))? ~((0x1L<<(n))-1) : 0 ))"
but i'm far from a good C programmer.

trinamic-LH, 2020-02-17:
Hello,

yes, the Arduino's int datatype has only 16 bits.

Am i understanding you correctly, that the CAST_Sn_TO_S32 macro is showing the bigger-than-16 shift warning?

PMTaylor, 2020-02-17:
Yes a warning is show. More importantly, the shifts fail to produce the correct result with the macro as written. Explicitly defining the type of the "1" and "0x1" constants as "uint32", or "unsigned long" here seems appropriate.

trinamic-LH, 2020-02-18:
So something like this should work then:
`#define CAST_Sn_TO_S32(value, n) ((value) | (((value) & ((uint32_t)1<<((n)-1)))? ~(((uint32_t)1<<(n))-1) : 0 ))`

Once we've tested everything on our end we will integrate that fix. Thanks for the report 👍

PMTaylor, 2020-02-18:
Yep, that works for me, hopefully it doesn't break anyone else's use cases.
I think we know the shift will never be more than 32 bits, so uint32 is the
right choice.

greyltc, 2020-06-07:
This looks like the same compiler warning I'm getting today:
```
Compiling src/TMC5130/TMC5130.c.o
src/TMC5130/TMC5130.c: In function 'tmc5130_writeDatagram':
src/TMC5130/TMC5130.c:22:22: warning: left shift count >= width of type [-Wshift-count-overflow]
  int32_t value = (x1 << 24) | (x2 << 16) | (x3 << 8) | x4;
                      ^
src/TMC5130/TMC5130.c:22:35: warning: left shift count >= width of type [-Wshift-count-overflow]
  int32_t value = (x1 << 24) | (x2 << 16) | (x3 << 8) | x4;
```
Would you agree?

trinamic-LH, 2020-06-10:
Fixed in commit b17de70de73c691f3cf71c5ff52a85527f5e3a7c

closed 2020-06-10 by trinamic-LH