analogdevicesinc/TMC-API #4: TMC5160 and Raspeberry PI

thalesfsp, 2020-04-21:
Hi! I'm trying to follow this [blog post](https://blog.trinamic.com/2019/03/29/internet-of-moving-things-tmc5130-raspberry-pi-3b/), but applied to the TMC5160 BOB. I cloned the latest version of the API, and made a few adjustments in the code, but I'm having problems with the `init` function.

The signature presented in the blog post, `void TMC5130_init(unsigned char motor)` is different from the API `void tmc5130_init(TMC5130TypeDef *tmc5130, uint8_t channel, ConfigurationTypeDef *config, const int32_t *registerResetState)`

I've limited knowledge on C, and in electronics. I searched around the web for examples of this new signature, and how to use/call that, but unfortunally did not find.

Thank you!

Lionheart1810, 2020-04-21:
Hi,
the source code for the Raspberry 3B has, per default, nothing to do with the TMC-API. It is a coherent project, hence the different functions and function signatures. So you have to decide if you want to use the functions presented there or if you want to switch to using the TMC-API, which is possible of course.

If you want to use the TMC-API:

The `TMC5160TypeDef` is just a structure for storing all neccessary information needed by the API (for e.g. register shadowing). You can look at it like a logical instance of an IC. You can look inside this structure, but you dont need to. The API takes care of it. Same with ConfigurationTypeDef instances.

So what you have to do is basically create a instances of `TMC5160TypeDef` and `ConfigurationTypeDef` (one each for every individual IC) somewhere in your project.c file, pass both as pointers to the tmc5130_init function and give it a channel number. You can assign these numbers arbitrarily, they are used to differentiate between communications channels later on. If you just have one channel, in this case one SPI channel, you can just use 0.
As `registerResetState` you can just use `tmc5160_defaultRegisterResetState` provided by the TMC-API.

After that, just implement the function `tmc5160_readWriteArray(...)`, which is needed by the API to read / write using your SPI.

A reference implementation for the TMC5160-EVAL can be found [here](https://github.com/trinamic/TMC-EvalSystem/blob/master/boards/TMC5160_eval.c). This can be applied to the TMC5160-BOB similarly.

thalesfsp, 2020-04-24:
Hello, @Lionheart1810! Thank you for your help! Can you help one more time providing me example of instances for both `TMC5160TypeDef` and `ConfigurationTypeDef?` What is required to be defined in these both instances? Finally, I would like to learn more about it. Would you be available for 30mins to join a chat and review my case? Cheers!

Lionheart1810, 2020-04-24:
Example instances are already in the linked reference implementation. They are created as simple as
```C
ConfigurationTypeDef config;
TMC5160TypeDef TMC5160;
```
After creation, you don't actually have to fiddle with its contents at all. Generally speaking, `ConfigurationTypeDef` offers state information and register shadowing to your IC instance.
Also, remember to take care of some registers you might need to set to certain values to get your specific setup running.
Yes I guess I can take some time to look over it. Which platform though?

thalesfsp, 2020-04-27:
@Lionheart1810 I have Keybase, Zoom and Google Meet. You choose. How can I DM you? Email? Cheers and really thank you for your time and help, I appreciate that! Cheers!

Lionheart1810, 2020-05-04:
Send me an E-Mail to kugis@trinamic.com

thalesfsp, 2020-05-06:
Thank you! I will reach you! Cheers!

closed 2020-05-18 by thalesfsp

thalesfsp, 2020-06-13:
Hello @Lionheart1810! I did message you :)