analogdevicesinc/TMC-API #8: Permissions descrepency

riggs, 2020-07-10:
https://github.com/trinamic/TMC-API/blob/077adb2820f6f907878ef65ccd4ba6ca35489dd2/tmc/ic/TMC5160/TMC5160.h#L75 shows `0x3D` as reserved, but that's actually write-only `ENC_DEVIATION`.  I'm guessing this is an error.

riggs, 2020-07-10:
I'm also noticing that a number of buffers (GSTAT, RAMPSTAT, & ENC_STAT) are marked with `0x21: read, flag register (read to clear)`, when they're in fact *write* to clear flags. I would think these should be `0x23`, possibly even with the clear bits set in the `tmc5160_defaultRegisterResetState` so the flags are cleared on reset.

trinamic-LH, 2020-07-10:
The register permissions are fixed in 3f5aa4c32bea6615a785a029a2ed21688e0e1506

As for clearing bits on reset - we don't do that on purpose. We want the API to have as few side effects as possible, including not clearing flags automatically.

closed 2020-07-10 by trinamic-LH