analogdevicesinc/TMC-API #14: documentation

qiuwenhui, 2021-02-23:
Is there documentation available?

miracuruzrh, 2021-04-22:
Would also appreciate a documentation

ernestguillem, 2021-04-23:
Documentation or some examples would be really useful. Thanks a lot in advance.

trinamic-LH, 2021-04-26:
Hello everyone,

I just added some first rudimentary notes to the README.
Lack of documentation is sadly a known issue.

I will check how quickly we can get more documentation written - in the meantime feel free to ask questions in this issue. I will answer and add the answers to the existing notes.

Regards,
Lenard

miracuruzrh, 2021-05-13:
> Hello everyone,
>
> I just added some first rudimentary notes to the README.
> Lack of documentation is sadly a known issue.
>
> I will check how quickly we can get more documentation written - in the [...]

Thank you very much.
'm using the API with a TMC5130_BOB and a teensy4.1 using VSC with platformio.ini.

So i included the API and my Program looks like this
`#include <Arduino.h>
#include <SPI.h>

#include <tmc/ic/TMC5130/TMC5130.h>
#include <tmc/ic/TMC5130/TMC5130.c>
#include <tmc/helpers/API_Header.h>

TMC5130TypeDef * myTMC;
ConfigurationTypeDef * myConfig;
uint8_t data[5] = { 0x21 | TMC5130_WRITE_BIT, 1, 1, 1, 1 };

// TMC5130TypeDef * myTMC;
// ConfigurationTypeDef myConfig;

void setup() {
    // myTMC->config->channel = 1;
    // myTMC->config * myConfig;
    tmc5130_init(myTMC, 1, myConfig, tmc5130_defaultRegisterResetState);
    tmc5130_writeInt(myTMC, TMC5130_VMAX, 30);
    // tmc5130_writeDatagram(myTMC,0x21,0x21,0x21,0x21,0x21);
    // tmc5130_readWriteArray(1,data,2);
    // tmc5130_reset(myTMC);
    // tmc5130_periodicJob(myTMC, 1);
}

void loop() {
    // tmc5130_periodicJob(myTMC,1);
 }`

You can see because of the commented out lines that I don't know exactly how to use the functions.
Wit the code in this status I get always some errors when I try to compile it.
Error message:
`Building in release mode
Compiling .pio/build/TMC_Original/src/main.cpp.o
In file included from src/main.cpp:5:0:
include/tmc/ic/TMC5130/TMC5130.c: In function 'void tmc5130_writeDatagram(TMC5130TypeDef*, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)':
include/tmc/ic/TMC5130/TMC5130.c:19:30: warning: narrowing conversion of '(int)(address | 128u)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
  uint8_t data[5] = { address | TMC5130_WRITE_BIT, x1, x2, x3, x4 };
                              ^
Linking .pio/build/TMC_Original/firmware.elf
.pio/build/TMC_Original/src/main.cpp.o: In function `setup':
main.cpp:(.text.setup+0x6): undefined reference to `tmc5130_readWriteArray(unsigned char, unsigned char*, unsigned int)'
collect2: error: ld returned 1 exit status
*** [.pio/build/TMC_Original/firmware.elf] Error 1`

I don't know now how to resolve this problem.
If you could help with a few basic instructions how to call the functions then I think I can adapt it and modify to my needs.
Help is much apprechiated.

clehanka, 2021-05-17:
@miracuruzrh There are Arduino libraries for the tmc ICs you maybe want to use. 
Regarding your error: You need to implement the tmc5130_readWriteArray function by yourself. (see usage in readme)
This library relies on you to write the hardware specific code which is important for portabillity. 
Since you use arduino you can make your life a little easier by using an arduino library.

trinamic-LH, 2021-05-17:
Hello,

I forgot to mention a function in the short overview - you must provide a wrapper function for the physical hardware (SPI, UART) to the TMC-API. This function is then called whenever the TMC-API wants to talk to the IC.

In the case of the TMC5130, you must implement the [tmc5130_readWriteArray](https://github.com/trinamic/TMC-API/blob/c5016fb2546c3e439f907d4733724e8227a7c2fb/tmc/ic/TMC5130/TMC5130.c#L10) function.

The function call sequence should be:
- Call tmc5130_init()
- Call tmc5130_periodicJob(). Initially, this will perform the autoconfiguration of the initial reset values step by step. Once this autoconfiguration is complete, the `myTMC->config->state` variable is set to `CONFIG_READY` [1]. Then you can start using read/writeInt functions. Note that the `myTMC->config->state` access is the only time you currently should ever look inside the TMCXXXXTypeDef and/or ConfigurationTypeDef objects (and we plan on adding proper wrapper functions for this).

closed 2022-03-25 by trinamic-ed

fabianmuehlberger, 2023-10-02:
Hello, 
any news on the state of the documentation ? I am currently working on a project using a ESP32 with ESPHome and the TMC2300 driver IC. And I am wondering if I can safe myself some time digging into the implementation. 
ESPHome could use the arduino framework, but I rather stick to the API, since it would allow a more stanardised approach and the ESP IDF would be the prefered framwork. 

The wrapper functions would be also very handy. 
> Note that the myTMC->config->state access is the only time you currently should ever look inside the TMCXXXXTypeDef and/or ConfigurationTypeDef objects (and we plan on adding proper wrapper functions for this).

Since I am relativly new to hardware specific programming, the code documentation would be appreciated. But I also would be happy to contribute, writing a framwork specifig library as well as documenting. 

Best, 
Fabian