analogdevicesinc/TMC-API #20: Is this a correct comparison?

ChrisIdema, 2021-11-23:
https://github.com/trinamic/TMC-API/blob/17d9adbc8cc3e3f2c7615ed194ad2518aa76fdae/tmc/ic/TMC429/TMC429.c#L334

if the first comparison is true the second one should be true as well so that one is never evaluated. I see the same comparison in the datasheet. Since pmul will be written to a 7-bit number shouldn't it be like so?:

`if((0 >= pmul) && (pmul <= 127)) `

trinamic-LH, 2021-11-23:
Hi,

The comparison in the code checks whether pmul is between 0 and 127 (both inclusive).
In the code, the check is correct. If the number is less than 0, the first condition fails (and the second one is not evaluated). If the number is greater than 127, the first condition succeeds, but then the second one fails.

Do note that the `(0 <= pmul)` comparison has the variable on the right hand side. It is equivalent to `(pmul >= 0)`. Maybe this caused the confusion?

closed 2021-11-23 by trinamic-LH

ChrisIdema, 2021-11-23:
> Do note that the (0 <= pmul) comparison has the variable on the right hand side
You are right. I missed that. Not very conventional to put the constant on the left, although I understand it is optically closer to `a <= x <= b`. I always have all warnings enabled so I never use assignment as comparison.