analogdevicesinc/TMC-API #21: platformio library

cooked, 2021-12-05:
I'm wondering if this is something in the pipeline? 
I've been using the TMC-API for quite some time now with the platformio environment and I like to keep the reference to the full repo tree (for the sake of keeping it up to date) instead of fishing out only the code I need. 
However doing it this way platformio always compiles all the files in the lib, needless to say throwing a lot of errors.

I've made a fork where I created a library.json so that using  the platformio library manager ( lib_deps = <myrepo> ) I get the desired behavior (see https://github.com/cooked/TMC-API/blob/master/library.json where I included only the files needed to work with the TMC2209)
The approach is not "flexible" yet, more of a proof of concept, so I'm wondering if there's interest in supporting it or someone else is facing similar situation.

trinamic-ASU, 2025-01-29:
We have started to rework TMC-API(more info is available [here](https://github.com/analogdevicesinc/TMC-API/issues/53)).

With the reworked implementation, one only needs to copy the folder of the IC they are using. For example in your case, you just need TMC-API/tmc/ic/TMC2209 folder.

BusHero, 2025-04-07:
Copying folders around is not a great development experience. PlatoformIO allows the installation of a dependency through the package manager, which is much nicer. 
It is relatively easy to create a PlatofrmIO library wrapper around TMC-API. You only need a repository with `library.json` in it with the following content:
```json
{
    "$schema": "https://raw.githubusercontent.com/platformio/platformio-core/develop/platformio/assets/schema/library.json",
    "name": "TMC PlatformIO",
    "repository": {
        "type": "git",
        "url": "https://github.com/BusHero/TMC-PlatformIO.git"
    },
    "authors": [
        {
            "name": "Name Surname",
            "email": "name.surname@example.com",
            "url": "https://github.com/repo"
        }
    ],
    "license": "MIT",
    "frameworks": "*",
    "platforms": "*",
    "build": {
        "includeDir": "TMC-API/",
        "flags": [
            "-ITMC-API/tmc/ic/MAX22216",
            "-ITMC-API/tmc/ic/TMC262",
            "-ITMC-API/tmc/ic/TMC2130",
            "-ITMC-API/tmc/ic/TMC2160",
            "-ITMC-API/tmc/ic/TMC2208",
            "-ITMC-API/tmc/ic/TMC2209",
            "-ITMC-API/tmc/ic/TMC2224",
            "-ITMC-API/tmc/ic/TMC2225",
            "-ITMC-API/tmc/ic/TMC2226",
            "-ITMC-API/tmc/ic/TMC2240",
            "-ITMC-API/tmc/ic/TMC2262",
            "-ITMC-API/tmc/ic/TMC2300",
            "-ITMC-API/tmc/ic/TMC2660",
            "-ITMC-API/tmc/ic/TMC4361A",
            "-ITMC-API/tmc/ic/TMC4671",
            "-ITMC-API/tmc/ic/TMC5031",
            "-ITMC-API/tmc/ic/TMC5041",
            "-ITMC-API/tmc/ic/TMC5062",
            "-ITMC-API/tmc/ic/TMC5072",
            "-ITMC-API/tmc/ic/TMC5130",
            "-ITMC-API/tmc/ic/TMC5160",
            "-ITMC-API/tmc/ic/TMC5240",
            "-ITMC-API/tmc/ic/TMC5262",
            "-ITMC-API/tmc/ic/TMC5271",
            "-ITMC-API/tmc/ic/TMC5272",
            "-ITMC-API/tmc/ic/TMC6100",
            "-ITMC-API/tmc/ic/TMC6200",
            "-ITMC-API/tmc/ic/TMC7300",
            "-Wno-discarded-qualifiers",
            "-Wno-unused-variable",
            "-Wno-pointer-sign",
            "-Wno-implicit-function-declaration"
        ]
    },
    "export": {
        "include": [
            "TMC-API/*"
        ],
    },
    "version": "3.10.10"
}
```
Here is an example on how I've done it:
https://github.com/BusHero/TMC-PlatformIO