analogdevicesinc/TMC-API #27: Documentation says user needs to implement wrong function for TMC4671

bigbrett, 2022-07-05:
The documentation in README.md states:

> 
> To use the TMC-API, perform the following steps in your code:
> - Implement the tmcXXXX_readWriteArray() function in in your code. This function provides the necessary hardware access to the TMC-API.
> 

But in the TMC4671 IC driver this function does not exist. Instead, it seems that the code wants you to implement [tmc4671_readWriteBytes()](https://github.com/trinamic/TMC-API/blob/master/tmc/ic/TMC4671/TMC4671.c#L15)

```
// => SPI wrapper
extern uint8_t tmc4671_readwriteByte(uint8_t motor, uint8_t data, uint8_t lastTransfer);
// <= SPI wrapper
```

Additionally, looking at other ICs, each function has a differently named wrapper implementation. This should be made uniform, or at least documented to the user so they know where to look (e.g. "Every IC has a function prototype marked as `extern` that the user needs to implement in order for the driver to communicate with the chip...` or something like that).

trinamic, 2022-07-12:
Ok, thanks! We will update the Readme file.

closed 2022-07-12 by trinamic

pwfff, 2023-06-04:
would be nice to see the README updated, confused the hell out of me too for the 2240