analogdevicesinc/TMC-API #32: Encoder Initialization by Hall sensors

Ayoub-pixe, 2023-05-03:
Hello @trinamic,

I was examining the tmc4671_doEncoderInitializationMode2 function and I came across the following line of code:

int16_t hall_phi_e_estimated = *hall_phi_e_old + tmc4671_getS16CircleDifference(*hall_phi_e_new, *hall_phi_e_old)/2;

I was curious as to why this equation is not applied to the encoder abn_phi_e_actual. Specifically, I'm wondering why we don't calculate an abn_phi_estimated using a similar formula:

abn_phi_e_estimated = abn_phi_e_old + (abn_phi_e_new - abn_phi_e_old)/2

Then we could set the TMC4671_ABN_DECODER_PHI_E_PHI_M_OFFSET register to hall_phi_e_estimated - abn_phi_e_estimated. Can you please explain why the offset average is only calculated for the hall sensor and not the encoder?

https://github.com/trinamic/TMC-API/blob/4ced0291c5cdae49f54e7f082c3c651d03ca19e7/tmc/ic/TMC4671/TMC4671.c#L406

trinamic-LH, 2025-04-02:
> Hella [@trinamic](https://github.com/trinamic) . I am working with [@Ayoub-pixe](https://github.com/Ayoub-pixe). We are asking this question because, we discovered that some encoder initialization were bad. Leading to a big error between phi-e and encoder. This big error leads to higher current in the motor (and the regulation) to drive it with the same load. We have identified that the high current is linked to offset error. So before trying to modify the Encoder initialization function. We wanted to understand why this equation was used. thanks.

For initializing an ABN, the choice in strategy matters a lot. A hall-based initialization as discussed here gives decent results only if:
- Your time between the hall edge occuring and performing the processing of the referenced C code happens quickly. Note that this includes the time to communicate with the TMC4671.
- There is no gearbox between the motor shaft and the ABN mounting point
  - Or if there is, little to no slack is present
  - Or if slack is present, it is well quantified and compensated (note the hysteresis-like effect of this, if the first hall edge is observed very quickly, the mechanical system may still not yet have fully started rotating)
- Your hall sensor and ABN sensor are decently mounted and/or mounting tolerances (hall edges not being at exactly 60° apart, ABN sensor not well aligned with the rotation axis) are being compensated for. 

For a more precise initialization, the ABN channel - if available - would be the best choice. Since the TMC4671 has the capability to directly process that input, no communication or processing delays are imposed like the hall. Since the N signal is part of the ABN itself, it will yield repeateable initialization. It does however require up to a full rotation of the mechanical axis to find the initially unknown N position.

Depending on your application, a multi-tier initialization strategy might also be feasible. Start with the current hall sector position estimate (+-30° error possible), then turn to identify an edge, lessening the error with at most 60° electrical rotation ( `60° / pole pair count` mechanical rotation). Then rotate until you found your ABN N channel (up to 360° mechanical rotation) to fully remove any ABN initialization error from the previous steps.

trinamic-LH, 2025-04-02:
(Feel free to reopen if there are any further questions)

closed 2025-04-02 by trinamic-LH