analogdevicesinc/TMC-API #34: default register values don't match datasheet

pwfff, 2023-06-16:
e.g. https://github.com/trinamic/TMC-API/blob/master/tmc/ic/TMC2240/TMC2240.h#L37

[image]

Are these defaults intended to match the reset values in the datasheet?

trinamic-ASU, 2023-06-21:
Thank you for reaching out to us. We appreciate your interest in our products/services.

The defaults used for the eval-board are typically not the same as the defaults of the IC after startup. The defaults of the eval-board should help to have good starting values with our eval-environment. Therefore, the default values in the FW could be different than the reset values mentioned in the datasheet.

Please let us know if you need further assistance.

pwfff, 2023-06-24:
I see, I thought this repository was separate from the eval board because it was meant to be generic. I think it would make more sense to have the chip's reset values in this repository, and the eval board's registers overridden in its repository. 

Either way, this code was very helpful so thank you for the work!

trinamic-ed, 2023-06-27:
Thanks for your infos, yes I think so too. We will split up the default: chip's reset in TMC-API and eval board register defaults in the TMC-EvalSystem repo.

closed 2023-06-27 by trinamic-ed