analogdevicesinc/TMC-API #40: Ramping examples?

slimcdk, 2023-10-08:
Hi there!

Is there examples of how to use the ramping implementation? 
I have been through a few different approaches, but can't seem to make it work correctly.

trinamic-ASU, 2023-10-11:
Hello,

Can you please specify which IC or eval-board you want to have the example for? 

Furthermore, please also elaborate what kind of ramping implementation you are looking for? Is it about six-point ramp?

Thank you for your interest in our products.

slimcdk, 2023-10-21:
Hello! I'm simply trying to understand how to use the linear ramping defined in tmc/ramp. I'm working on ESPHome integrations for the 2209, 2240 and 2300 driver IC and would like to utilize your ramping to create smooth movement. I'm currently only using the UART interface for control.

FYI, you can see my current work for the TMC2209 here https://github.com/slimcdk/esphome-custom-components/tree/master/esphome/components/tmc2209

trinamic-ASU, 2025-03-13:
You can take TMC4671 as reference [here](https://github.com/analogdevicesinc/TMC-EvalSystem/blob/9f38f12ce0699138c6971ae8339bf781a4beb3b5/boards/TMC4671_eval.c#L1070).

closed 2025-04-02 by trinamic-ASU