analogdevicesinc/TMC-API #43: Proposal: Create read the docs Documentation

fabianmuehlberger, 2023-10-17:
Hi, 
I would like to propose to create an official documentation using [Readthedocs](https://about.readthedocs.com/?ref=readthedocs.org).
This would certainly help others working with your API and bring more contributions to this platform. 

Why readthedocs (rtd) ? It is widely used for documenting projects and is highly customizable and works with multiple languages.

I created [a basic example](https://tmc-api.readthedocs.io/en/latest/source/TMC2300.html) how the TMC-API documentation could look like, using rtd, sphinx doxygen for generating the documentation. 
This workflow allows automated generation of documentation based on the doxygen comments in the code itself and therefore version controlled docs. Additionally, sphinx is used to write more descriptive with reStructuredText than plain doxygen comments, and can create a more user-friendly documentation. 

The [documentation ](https://tmc-api.readthedocs.io/en/latest/source/TMC2300.html) I made is mostly autogenerated with a few doxygen styled comments for a few functions for testing, but this is nearly fully autogenerated and can be generated by clicking one button or automated via GitHub actions. 

I hope this idea is of interest to you and helps develop this amazing hardware platform!

Best, 
Fabian

trinamic-ed, 2023-10-18:
Hello Fabian,

we discussed the documentation topic internally. 

E.g. for our iolink python package, we made a readthedocs.io [doku](https://iolink.readthedocs.io/en/latest/) some time ago. 

We've seen that you started a documentation using doxygen. In the moment it seems to be a copy of the C-code function header, there must be added more to make it more usable.

In the moment we do not focus on the doku. We will upade the source code in the next weeks/month to make it easier usable also for Arduino and Raspi, so the TMC-API code will change in the next time and also the use inside the TMC-EvalSystem.

Best regards,
    Enrico

fabianmuehlberger, 2023-10-18:
Hey, 
Thank you for the reply, these are good news. As I mentioned in my proposal, the work I did was purely a proof of concept since I had no prior experience with readthedocs. 

Since I had no intention to change the code base or just doxygen comments I left it fully autogenerated with some test how to structure the doc with .rst files. 

I would also prefer a documentation that is not only the autogenerated doxygen but a more descriptive implementation with examples and helpful information as well as explaining the overall concept of the API. 

I will take down the readthedocs again, since it is misleading. 
If I can help in any way, let me know, I am happy to contribute. Especially because I am working on an implementation using the TMC-API for ESPhome using the esp-idf. 

Information about the upcoming changes would be very appreciated, since 2 people are currently putting a lot of work in it!

trinamic-ASU, 2025-01-29:
Hi Fabian,

We have started to rework TMC-API (as mentioned by @trinamic-ed). You can find more information [here](https://github.com/analogdevicesinc/TMC-API/issues/53).

If you have any questions or suggestions, please let us know. We highly appreciate your feedback and interest in our products!