analogdevicesinc/TMC-API #46: Issues with reading XACTUAL from TMC5160

hajilisamir, 2023-11-10:
Hi,

I have a situation that when I write 0x05 0x07 0x21 0x9E data package to my stepper driver through UART I read nothing. I confirm this through Logic Analyzer.
[image]

Any ideas?

Thanks in advance
Samir

joshua-8, 2023-11-10:
Hi Samir,
I've never used a TMC5160, but looking at the datasheet in section 23.1 Initialization Examples, it suggests that you can "query XACTUAL" with `UART write: 0x05 0x01 0x21 0x6B` so that's probably worth trying. How did you choose the 4 bytes in your data package?
Good luck,
joshua-8

hajilisamir, 2023-11-10:
Hi Joshua,

Thank you for your reply.

It is the similar message. 0x07 is the address instead of the 0x01 from the example. It is different since I have multiple steppers in my set up.

To give you more details about the issue; I have this messages sent in the past and I was able to read the signals correctly with Logic Analyzer, and since then I was working both on HW and SW, now something is up with which I don't understand, so steppers do not reply back. Could be a hardware issue. Here is an example when the stepper driver replied to the same 4 byte data package.

[image]

Till I confirm if it is a HW issue or not, do you have any clue what would be going wrong?

Thank you and best regards,
Samir

trinamic-ASU, 2025-03-13:
We have reworked TMC-API. Please use the latest code and if the problem remains, please let us know.

Thank you for your feedback.