analogdevicesinc/TMC-API #53: Draft: TMC-API structural migration #1

trinamic-ASU, 2024-02-27:
The TMC-API is getting a rework of its core structure. This is to make the direct accessing of chip registers as easy as possible.

For detailed descriptions of how to use the updated TMC-API, see here: [README](https://github.com/analogdevicesinc/TMC-API/master/tmc/ic/TMC5272/README.md)

# Migration step 1: Add new register access method
Migration progress:
- [x] TMC5272
- [ ] TMC5271 (WIP)
- [ ] TMC2209 (WIP)
- [ ] TMC5240 (WIP)

# Migration step 2: Move caching, reset and restore logic to TMC-EvalSystem
- [ ] TMC5272 (WIP)

If you'd like us to prioritize a specific chip feel free to comment in this issue

robw-ils, 2024-04-02:
> If you'd like us to prioritize a specific chip feel free to comment in this issue

TMC5130 would be great. Thank you!

trinamic-ASU, 2024-04-03:
> > If you'd like us to prioritize a specific chip feel free to comment in this issue
> 
> TMC5130 would be great. Thank you!

Thank you for bringing your request to our attention. Rest assured, we have noted it down. 
We appreciate your feedback.

F14V, 2024-07-06:
Hi! Could you please also add the TMC2240 to the list? I see that TMC5240 is included, but not the former.

trinamic-ASU, 2024-07-07:
> Hi! Could you please also add the TMC2240 to the list? I see that TMC5240 is included, but not the former.

Thank you for bringing your request to our attention. Rest assured, we have noted it down.
We appreciate your feedback.

Sbahuddin1, 2024-07-20:
Please do TMC4671 as well. and how long will it take? thanks. 

I can't see it in the list above.

trinamic-ASU, 2024-08-02:
> Please do TMC4671 as well. and how long will it take? thanks.
> 
> I can't see it in the list above.

Thank you for bringing your request to our attention. We have duly noted your request; however, it may take some time to address it. As we have a significant number of other ICs in the pipeline and already under progress. 

Rest assured, we understand the importance of your request and will prioritize it as soon as possible. 

If you have any further questions or concerns, please do not hesitate to reach out to us. Thank you for your continued support.

slimcdk, 2024-08-15:
Would like to see TMC5240 be reworked soon! I just updated my work for the TMC2209 and would like to continue on the TMC5240.

trinamic-ASU, 2024-08-16:
> Would like to see TMC5240 be reworked soon! I just updated my work for the TMC2209 and would like to continue on the TMC5240.

Sure!
It would be released by the end of next week.
Thank you for bringing your request to our attention.

CupeTheProgammer, 2024-08-22:
Hello, Ive commited to use TMC4671-BOB and TMC6200-BOB with my STM32F030x8 which I am programming in STM32CubeIDE for self education. Would like to see both TMC ICs migrated soon. 

Tbh I am confused with current API as I am newbie to programming. Is there some more detailed guide to work with the current API other than evalsystem example codes? Or should I just wait for it to be migrated?

trinamic-ASU, 2024-08-23:
> Would like to see TMC5240 be reworked soon! I just updated my work for the TMC2209 and would like to continue on the TMC5240.

As promised, API rework for TMC5240 is ready and has been published on the master branch.

slimcdk, 2024-08-23:
> > Would like to see TMC5240 be reworked soon! I just updated my work for the TMC2209 and would like to continue on the TMC5240.
> 
> As promised, API rework for TMC5240 is ready and has been published on the master branch.

Thanks! Would also like to see TMC2300 migrated at some point

fabianmuehlberger, 2024-09-01:
I would also appreciate to see the TMC2300 being migrated. 

Best

hnvseau, 2024-09-18:
My company is using TMC 4671 and we would like it to be migrated

trinamic-ASU, 2024-09-25:
> Hi! Could you please also add the TMC2240 to the list? I see that TMC5240 is included, but not the former.

Hi,
TMC2240 has been released on the master branch.

trinamic-ASU, 2024-09-25:
> > > Would like to see TMC5240 be reworked soon! I just updated my work for the TMC2209 and would like to continue on the TMC5240.
> > 
> > 
> > As promised, API rework for TMC5240 is ready and has been published on the master branch.
> 
> Thanks! Would also like to see TMC2300 migrated at some point

Hi, 
TMC2300 has been released on the master branch.

trinamic-ASU, 2024-10-16:
> Hello, Ive commited to use TMC4671-BOB and TMC6200-BOB with my STM32F030x8 which I am programming in STM32CubeIDE for self education. Would like to see both TMC ICs migrated soon.
> 
> Tbh I am confused with current API as I am newbie to programming. Is there some more detailed guide to work with the current API other than evalsystem example codes? Or should I just wait for it to be migrated?
> 
> Photo of the process of learning. ;) ![IMG_1993](https://private-user-images.githubusercontent.com/179036477/360557512-41b844c9-091b-4ac2-9c0f-edf990e81ca2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkwNzc1ODYsIm5iZiI6MTcyOTA3NzI4NiwicGF0aCI6Ii8xNzkwMzY0NzcvMzYwNTU3NTEyLTQxYjg0NGM5LTA5MWItNGFjMi05YzBmLWVkZjk5MGU4MWNhMi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxNlQxMTE0NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ZjE1NWNiN2ExMTE2NmQ5MTM0NmJiZGY2MGJjMjYyYzA0NGEyOTVkNjgwZWE5NTE5ODNlNjQ0ZDlmOWUwNjI0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.j3PpjHnWV_3od56LWbsTd2eiAqNdb-G3dpHJAWvC6co)

TMC4671 has been released on the master branch.

trinamic-ASU, 2024-10-16:
> My company is using TMC 4671 and we would like it to be migrated

Its done!

grahas, 2025-02-01:
TMC4361A would be great

trinamic-ASU, 2025-02-11:
> TMC4361A would be great

It's done and has been released on master branch.

grahas, 2025-02-11:
> > TMC4361A would be great
> 
> It's done and has been released on master branch.

🤩 Thank you so much!

closed 2025-03-13 by trinamic-ASU