analogdevicesinc/TMC-API #57: Some definitions are missing in TMC5272_HW_Abstraction.h compared to the data sheet

chrislro2019, 2024-04-12:
I did not do a full review, but I stumbled over missing definitions in the header file compared to the data sheet. The header file should contain for all registers all field definitions as defined in the data sheet. 

E.g. 
https://github.com/analogdevicesinc/TMC-API/blob/f276b3e94894494009a1c44e3eb21512ad0ab320/tmc/ic/TMC5272/TMC5272_HW_Abstraction.h#L520-L526

lacks the definition for the fields:
- qsc_enc_en
- BEMF_FILTER_SEL

In addition - at least for the TMC5272 - the header file does not define field values as defined in the "Decode" column in the data sheet. This would be very helpful to have all of them in the header file as there are really a lot of them. E.g.: 

```
#define TMC5272_ENCMODE_POL_B_NEGATIVE (0)
#define TMC5272_ENCMODE_POL_B_POSITIVE (1)
```

chrislro2019, 2024-04-12:
Also a goody would be the list of all register RESET values as defines.

PMTaylor, 2025-12-12:
also missing appears to be the external Encoder tolerance, P/gain value, and tolerance enable/disable fields.