analogdevicesinc/TMC-API #59: License Clarification

DaAwesomeP, 2024-06-12:
Hello,

This repo appears to have an MIT license, but some files contain text like the following: https://github.com/analogdevicesinc/TMC-API/blob/9dbaf83d433b3bb63b784249f716c685e4c4aced/tmc/ic/TMC5240/TMC5240_Fields.h#L1-L7

Surely it isn't confidential because it is on GitHub, and this text also contradicts the MIT license. May these files be updated to reflect the correct licensing?

Thanks!

trinamic-ed, 2024-06-13:
Hello,

you are pointing to the copyright which is inside all files. The license is located [here](https://github.com/analogdevicesinc/TMC-API/blob/master/LICENSE) and valid for all files.

> this text also contradicts the MIT license. May these files be updated to reflect the correct licensing?

How should it look like from your view?

Regards,
    Enrico

trinamic-ed, 2024-06-13:
We will remove the "and confidential" in all the files in the next days.

DaAwesomeP, 2024-06-14:
> How should it look like from your view?

I'm not a lawyer, but the copyright notice here as written is more than just a copyright notice here. Obviously, "confidential" should be removed. But "proprietary" is also contradictory. The MIT license says:

> "without restriction, including without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of the Software, and to permit persons to whom the Software is furnished to do so..."

The MIT license says "without limitation" and "to deal in the Software without restriction," but "proprietary" means that all rights and ownership are _exclusive_ to the owner. This is different than "all rights reserved" which means that they are _reserved until granted_ and not exclusive to the owner. They are then granted by the MIT license.

An MIT license-compatible copyright notice would simply be: "Copyright (c) 2023 Analog Devices, Inc. All rights reserved."

closed 2025-01-29 by trinamic-ASU

DaAwesomeP, 2026-03-19:
Hi, may this please be reopened? This license header text contradicts the MIT license found in the repo. Either the license needs to change or the header text needs to change. Thanks!

schweigstill, 2026-03-22:
> Hi [@trinamic-ed](https://github.com/trinamic-ed),
> 
> Thanks for reopening this. Please see my notes above. This has now come up a few times in other open source projects trying to make use of TMC-API (example: [zephyrproject-rtos/zephyr#105388 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/105388#issuecomment-4086554833)).
> 
> The word "proprietary" is contradictory to "without restriction." The word "proprietary" implies that Analog Devices requires some legal control of the code that is at odds with the MIT license. Even if there is some technicality that allows this wording to remain, it effectively prevents the TMC-API from being used because it is confusing (and developers are not lawyers).
> 
> May you please clarify the meaning of "proprietary" in this context? How is this MIT-licensed code proprietary to Analog Devices?

It's considerably worse, or rather, more complicated: all these considerations must not only be made from the perspective of one legal system, but, especially in the case of globally used projects, must be considered separately for each relevant legal system. In this specific case, the German company Trinamic GmbH, the original rights holder, was also acquired by an US company. And any explanations provided by an employee whose specific responsibilities within the rights-holding company are not entirely clear to third parties are, in case of doubt, legally irrelevant anyway.

trinamic-ed, 2026-03-25:
Hello,

at the end it is not so complicated. I checked the copyright info with our license team and we will remove the part "_All Rights Reserved. This software is proprietary to Analog Devices, Inc. and its licensors._" in general. This leads to a shorter license text like e.g. this (instead of the example above):

/******************************************************************************* 
 * Copyright © 2019 TRINAMIC Motion Control GmbH & Co. KG 
 * (now owned by Analog Devices Inc.), 
 * 
 * Copyright © 2023 Analog Devices, Inc.
 *******************************************************************************/ 

Thanks for bringing up this topic!

Do you have any further concerns regarding the copyright text? Otherwise, we would start to update the copyrights for TMC-API and TMC-EvalSystem.

Regards,
     Enrico